SonarSonic/DrawingBotV3 #2: GCode - Enhancements/Bugs

SonarSonic, 2021-03-22:
- Add custom GCode commands
- Add flip / rotation options
- Bug: Offsets being ignored

reven, 2021-03-22:
This would be much much appreciated! While playing around with the gcode I also noticed that it generates an x,y move and then goes to "pen down" height if it is set, as bellow:
```
(GCode generated by: DrawingBotV3 1.0.7)

(Time: 2021-03-22 at 16:33:13 PDT)

G21
G90
G28
G1 F8000
G1 X2.183 Y82.986
G1 Z10.000
G1 X3.472 Y82.738
G1 X3.720 Y82.540
...
```
It should raise the pen to the "pen up" value if set before attempting any move. like so:

```
...
G1 F8000
G1 Z15.000
G1 X2.183 Y82.986
G1 Z10.000
...
```

This is a must for printers/plotters where the resting state of the head is at 0,0,0.

closed 2021-04-01 by SonarSonic

reven, 2021-04-05:
This is awesome! Thanks!! Can't wait to see this in release.

commit b342ef9a (2021-04-01): GCode Export - Improvements + Fixes #2
commit a22e406b (2022-01-19): GCode Export - Improvements + Fixes #2
commit efe09ce7 (2022-01-19): GCode Export - Improvements + Fixes #2