SonarSonic/DrawingBotV3 #4: G

reven, 2021-04-09:


SonarSonic, 2021-05-04:
Additional ones to fix when re-visiting g-code export.
- Drawing not being centred
- Check CMYK plots all have the same centre.

SonarSonic, 2021-05-07:
@reven What slicer are you using to preview the gcode?

closed 2021-05-07 by SonarSonic

reven, 2021-05-07:
> @reven What slicer are you using to preview the gcode?

Repetier host, though you have to mark the option to see movements.

SonarSonic, 2021-05-08:
@reven Let me know how you get on with the new version I've just released, hopefully it should fix all the issues mentioned above :)

reven, 2021-05-21:
Thanks @SonarSonic !!

Sorry I didn't get back to you sooner. I just did some export tests and the generated gcode seems perfect. Offsets are spot on, and everything looks good.