SonarSonic/DrawingBotV3 #24: Text Cloud

smmsamm, 2021-12-02:
Please add nesting text in a shape like text clouds

SonarSonic, 2021-12-03:
Could you provide an example of exactly what you mean by this?

smmsamm, 2021-12-03:
I searched and found this:
https://wordart.com/czzbhv6r8ym5/blackpink-jisoo-kim
you can use create button at top to find many features

SonarSonic, 2023-03-11:
Adaptive Letters was added in v1.5.2 - though this doesn't allow text only characters/text, so I'm keeping this open for now.

smmsamm, 2023-03-22:
thank you Please add these features too:
*Character Outlines color
*Character Fill Color
*Fill text characters by colors of input image(just under the position of characters)
*Checkbox Allow Overlapping Texts over each others or not
*Checkbox Allow put text inside others; For Example B can be inside a bigger O yes or not
*Select the font by typing the name
*Using Words too, Now it uses characters only
*Number of Repeatable characters, For example I want App uses only O 3 times not more
*Please have a look to:
http://eng.e-cut.ru/index.php?view=function&functionid=0
https://svgnest.com/

SonarSonic, 2023-03-22:
Some great ideas, some may not be possible but others will hopefully make it in!