SonarSonic/DrawingBotV3 #35: Index 0 out of bounds for length 0 on save of new Mosaic PFMs

barnyrelph, 2022-01-01:
First up, I absolutely love this software, it's wonderfully powerful and I'm getting some awesome results. Thanks very much for sharing it!

I tried out the Mosaic Rectangles and Mosaic Voronoi PFMs (both with default settings) and although the result on screen looks good, I get the error on save. Other PFMs seem to work fine. I'm on a MacBook Pro (Intel) if that helps? 

I've run the App from the console and grabbed the Stack Trace if that's of use? 
```

Jan 01, 2022 8:42:54 PM drawingbot.files.ExportTask setException
SEVERE: Export Task Failed
java.lang.IndexOutOfBoundsException: Index 0 out of bounds for length 0
	at java.base/jdk.internal.util.Preconditions.outOfBounds(Preconditions.java:64)
	at java.base/jdk.internal.util.Preconditions.outOfBoundsCheckIndex(Preconditions.java:70)
	at java.base/jdk.internal.util.Preconditions.checkIndex(Preconditions.java:248)
	at java.base/java.util.Objects.checkIndex(Objects.java:372)
	at java.base/java.util.ArrayList.get(ArrayList.java:459)
	at drawingbot.geom.spatial.STRTreeSequencer.sequence(STRTreeSequencer.java:71)
	at drawingbot.geom.spatial.STRTreeSequencer.sort(STRTreeSequencer.java:92)
	at drawingbot.geom.GeometryUtils.getOptimisedGeometriesForExport(GeometryUtils.java:127)
	at drawingbot.geom.GeometryUtils.getGeometriesForExportTask(GeometryUtils.java:54)
	at drawingbot.files.ExportTask.call(ExportTask.java:95)
	at drawingbot.files.ExportTask.call(ExportTask.java:22)
	at javafx.concurrent.Task$TaskCallable.call(Task.java:1425)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)
```

barnyrelph, 2022-01-01:
Having looked at the stack trace for a minute, I've realised that if I disable optimisations, the save is successful and I can get files saved, so that's a useful workaround for now :)

Kenno9000, 2022-01-01:
On 1.2.2, this error shows up quite frequently, in both lines and curves (that I use the most).  There appear to be a few parameters that generate the issue, and a solution can be found (often, not always) but there is no one path to the solution.  This might be self-evident, but the issue occurs only upon trying to export (all modes are affected).

SonarSonic, 2022-01-01:
@barnyrelph Happy to hear your enjoying DBV3!

This will be fixed in the next version and is a duplicate of #31

closed 2022-01-01 by SonarSonic