SonarSonic/DrawingBotV3 #50: Overlapping Circles Style

SonarSonic, 2022-03-12:
As per email suggestion.
[image]

papiend, 2022-03-13:
It would be great if thescribbler could output a path in .svg so that it could be used in a drawing machine.
If anyone could help, I would be grateful forever. 

[](http://3dstereophoto.blogspot.com/2019/04/painting-software-scribbler.html)

papiend, 2022-03-13:
[http://3dstereophoto.blogspot.com/2019/04/painting-software-scribbler.html](url)

SterkArtLine, 2022-03-21:
@papiend  Yes, we are in the same situation. I've searched too, but someone with good coding knowledge needs help. I hope a solution is found.

SonarSonic, 2022-04-18:
[image]

I'm working on it ;)

SterkArtLine, 2022-04-18:
You're amazing Ollie, I'm so glad to see this message.  It was exactly the algorithm and drawing style I needed to “create gradations without multiple pencils”.  I'm counting on you good luck :)

led, 2022-04-18:
This looks fabulous. Itching to give it a 'whirl'.

papiend, 2022-04-20:
https://www.youtube.com/watch?v=vxjM5HyqyUI&list=PLmm_L3InpMwE6SbfY1IE17f0Zcmc-WbGd&index=3

Maybe this is where you can get an idea ...

SonarSonic, 2022-04-22:
[image]

SterkArtLine, 2022-04-22:
The option to have lines only in the portrait, without the lines in the background like other filters in DrawingBot software, would be great.

SonarSonic, 2022-04-22:
@SterkArtLine This isn't the final version of the Path Finding Module, I'm trying to get it as close as possible to the paper [_Tone- and Feature-Aware Circular Scribble Art_](https://www.researchgate.net/publication/282940942_Tone-_and_Feature-Aware_Circular_Scribble_Art) as possible currently and in their implementation they don't limit the curve areas in white areas of the image as this can lead to artifacts. I do think it will be possible though, but it may be that it only works really well on images like he one above, where there is a clear subject.

SonarSonic, 2022-04-22:
I'll reopen this issue actually as an for enhancements to the Algorithm

- [ ] Option to prevent the curves from drawing over white areas of the image.

SterkArtLine, 2022-04-22:
[image]

Here is an example of a picture I drew with the ChiuEtAl algorithm earlier. There are no lines in the white areas. It's the same thing we're talking about. I eagerly await the results. I hope it ends soon, Ollie. I wish you ease.

SonarSonic, 2022-04-22:
Yeah that's style is the aim. Where can I find this implementation of the Algorithm to compare results?

SonarSonic, 2022-04-22:
Ahh I found it. I've been coding mine from scratch, I shall make some comparisons.

SterkArtLine, 2022-04-22:
Well, I can send the file if you need it.

SterkArtLine, 2022-04-22:
Ollie Lansdell ***@***.***>, 22 Nis 2022 Cum, 16:42 tarihinde
şunu yazdı:

SonarSonic, 2022-05-03:
Just an update to say this algorithm was added in the latest version, you should use image with a high resolution and very high contrast for the best results and if you wish to achieve results similar to Chiu Et Al’s implementation use a size of paper, pen width which gives you a plotting size of 4000px on the largest edge then use the “Chiu Et Al – 4000px” preset.

papiend, 2022-10-18:
[image]

led, 2022-10-18:
To fix the pen lift issue, set your clipping mode to 'Page' (you'll find it in the 'Drawing Area' panel). You might need to make sure you've got a slight margin around your image too.

SonarSonic, 2023-03-11:
Closing as Adaptive Circular Scribbles is out of Beta as of v1.5.2 and producing really lovely results!

closed 2023-03-11 by SonarSonic