SonarSonic/DrawingBotV3 #77: Application doesnt open on MacOs Ventura

SonarSonic, 2022-11-19:


JackWH, 2022-11-19:
This issue seems to be resolved. Ventura apparently changed the warning dialog which appears when opening an unsigned application, and added a second layer of confirmation too.

The app can be opened by going to System Preferences --> Privacy and Security, then clicking "Open Anyway". The OS will give a second error, but returning to System Preferences and following the same process again will force it to open.

SonarSonic, 2022-12-08:
Thanks @JackWH - another fix for people with this issue is allowing apps from more sources.

[image]

SonarSonic, 2023-02-13:
Fixed in v1.5.2

closed 2023-02-13 by SonarSonic