SonarSonic/DrawingBotV3 #78: export animation

smmsamm, 2022-11-23:
1: an option for adding some frames at the end for make pausing some seconds;
for example I add 10 for pause and DB adds 10 last frame to make video pause at the end when playing
2: add reverse mode to the end for making infinity loop;
for example start from blank and draw to make complete drawing, pause in middle for some seconds, reverse draw and clear the canvas to blank

SonarSonic, 2022-11-24:
Option 1: is already available via "frame hold"
Option 2: is a good idea!

smmsamm, 2023-03-22:
Can App Import a Mp4, Process all frames(one by one) and save to a single Mp4 again with all processed Frames?

SonarSonic, 2023-03-22:
Currently there is not Video -> Video, you can do Video -> Individual Images per frame, but I think Video -> Video would be useful!