SonarSonic/DrawingBotV3 #81: Crash using color matching

dkmaj89, 2022-12-01:
**Describe the bug**
Program crash after selecting drawing area to other than original size and selected color match.

**To Reproduce**
Steps to reproduce the behavior:
1. Add any image
2. Change Drawing Area > Original Sizing to any other
3. Change Pen Settings > Colour Separation to color match
4. Press Start

**DrawingBotV3 Version**
Premium v1.5.0-beta

**Operating System**
Windows 10

SonarSonic, 2022-12-08:
Hey @dkmaj89,
I can't seem to recreate this issue, I have a few questions which may help track it down.
- Did this happen to you in v1.4.2?
- Is this happening with any PFM?
- What resolution is your input image?
- How much RAM do you have?

Thanks :)

SonarSonic, 2022-12-08:
Can you make it crash once more and then close the application and send your log files "prev_log.txt" and "latest_log.txt"

You can find them by going to Help / Open Configs Folder.

dkmaj89, 2022-12-08:
[latest_log.txt](https://github.com/SonarSonic/DrawingBotV3/files/10185023/latest_log.txt)
[prev_log.txt](https://github.com/SonarSonic/DrawingBotV3/files/10185024/prev_log.txt)

dkmaj89, 2022-12-08:
I noticed that prev_log hasn't been updated for a long time. In this message You can find lastest_log from crash
[latest_log.txt](https://github.com/SonarSonic/DrawingBotV3/files/10185167/latest_log.txt)

SonarSonic, 2022-12-15:
@dkmaj89 Try the new version, there have been some Colour Match fixes, hopefully they'll fix your issue

dkmaj89, 2022-12-17:
Tested didn't help.
Normaly i work on windows version but for test i run java version. After test on jar file I get some error that may help:

Command: java -jar DrawingBotV3-Premium-1.5.1-beta-all.jar >out.txt 2>err.txt (files in attachment)

Also I find out something that mey be helpful on jar version. After changing "original size" to any other and just clicking in "rescale to pen width (mm)" ploting size shows max int value 2,147,483,647 x 2,147,483,647 no matter if the image is loaded or not. (file err2.txt)

I have installed:
Java version 8 update 351
JDK 17.0.5 x64

[out.txt](https://github.com/SonarSonic/DrawingBotV3/files/10251458/out.txt)
[err.txt](https://github.com/SonarSonic/DrawingBotV3/files/10251459/err.txt)
[err2.txt](https://github.com/SonarSonic/DrawingBotV3/files/10251460/err2.txt)

SonarSonic, 2022-12-17:
That may be the issue only Java 17+ is supported, Java 8 is not supported anymore.

dkmaj89, 2022-12-17:
At https://www.java.com/pl/download/ java 8 update 351 is the newest version

SonarSonic, 2022-12-17:
I generally recommend not running from the jar but running form the .exe or .zip instead, as they have the correct Java 17 built in. The old java website doesn't have the latest downloads you need Oracle.com.

https://www.oracle.com/uk/java/technologies/downloads/#java17

SonarSonic, 2022-12-17:
Would you mind screen shooting your Drawing Area settings panel when the plotting size is 2,147,483,647 x 2,147,483,647 - this could be a Java Version Issue too but may be something to do with the values inputted there.

dkmaj89, 2022-12-17:
[image]

dkmaj89, 2022-12-17:
INFO: Notification: Loaded Image File : d:\Programy\dbv3_1.5.1-beta\d7e564116712b8e192654e5e3a27.jpeg
gru 17, 2022 3:06:44 PM drawingbot.utils.DBTask setException
SEVERE: TASK FAILED
java.lang.IllegalArgumentException: Dimensions (width=2147483647 height=2147483647) are too large
        at java.desktop/java.awt.image.SampleModel.<init>(SampleModel.java:131)
        at java.desktop/java.awt.image.SinglePixelPackedSampleModel.<init>(SinglePixelPackedSampleModel.java:144)
        at java.desktop/java.awt.image.Raster.createPackedRaster(Raster.java:915)
        at java.desktop/java.awt.image.Raster.createPackedRaster(Raster.java:547)
        at java.desktop/java.awt.image.DirectColorModel.createCompatibleWritableRaster(DirectColorModel.java:1032)
        at java.desktop/java.awt.image.BufferedImage.<init>(BufferedImage.java:333)
        at org.imgscalr.Scalr.createOptimalImage(Scalr.java:2006)
        at org.imgscalr.Scalr.scaleImage(Scalr.java:2133)
        at org.imgscalr.Scalr.resize(Scalr.java:1694)
        at drawingbot.image.ImageTools.cropToCanvas(ImageTools.java:300)
        at drawingbot.image.format.FilteredImageData.applyCropping(FilteredImageData.java:229)
        at drawingbot.image.format.FilteredImageData.updateAll(FilteredImageData.java:179)
        at drawingbot.image.ImageFilteringTask.call(ImageFilteringTask.java:23)
        at drawingbot.image.ImageFilteringTask.call(ImageFilteringTask.java:10)
        at javafx.concurrent.Task$TaskCallable.call(Task.java:1426)
        at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:317)
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:577)
        at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:317)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1144)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:642)
        at java.base/java.lang.Thread.run(Thread.java:1589)

SonarSonic, 2022-12-17:
How are you running it now with the .exe or .zip?

dkmaj89, 2022-12-23:
I find out solution. I have i9-10850k that have 20 cores. I reduce cores using by java using command:

`-XX:ActiveProcessorCount=16`

And everything works perfect. All between 1 and 16 cores works fine. Setting 17 and more give crash.

Solution for other: 
add to file `app/DrawingBotV3-Premium.cfg` in section `[JavaOptions]` line:

`java-options=-XX:ActiveProcessorCount=16`

If you recive error reduce `16` to your max cores or lower

peckerdunn, 2023-03-20:
I have a similar fault. Version 1.5.3. Java version = 1.8.0_361 (installed today)
I import an image. 
Choose the pen set.
Choose the colour match option.
Press start.
DB shows a quick message like '... comparing pen...' and then it crashes.

[latest_log.txt](https://github.com/SonarSonic/DrawingBotV3/files/11021592/latest_log.txt)

dkmaj89, 2023-03-21:
Did You try adding that `java-options=-XX:ActiveProcessorCount=16` line to your config file?

File `app/DrawingBotV3-Premium.cfg` should look like dis:

```
[Application]
app.classpath=$APPDIR\DrawingBotV3-Premium-1.5.3-stable-all.jar
app.mainclass=drawingbot.LauncherPremium

[JavaOptions]
java-options=-Djpackage.app-version=1.5.3
java-options=-Dprism.verbose=true
java-options=-Dprism.maxvram=1G
java-options=-Xmx16G
java-options=-Dprism.maxTextureSize=0
java-options=-XX:ActiveProcessorCount=16
```
change `16` with yor max cores

peckerdunn, 2023-03-21:
Thanks for your response. I tried reducing the ActiveProcessorCount to 12, 8 and 4; didn't work. 
I tried starting it Java and  exported the err.txt file as suggested above.
[err.txt](https://github.com/SonarSonic/DrawingBotV3/files/11027730/err.txt)
Also I include a screen grab.

https://user-images.githubusercontent.com/1184790/226595981-119f7cdb-74f9-4771-a841-3684fd91f9c3.mp4

Hope this helps.

SonarSonic, 2023-03-21:
@peckerdunn What GPU & CPU do you have? Do you encounter this issue when not using colour match?

peckerdunn, 2023-03-21:
Everything else seems to work fine. 
Here's what I got off the sys info 
OS Name	Microsoft Windows 10 Pro
Version	10.0.19045 Build 19045
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Type	x64-based PC
System SKU	SKU
Processor	AMD Ryzen 5 3600 6-Core Processor, 3600 Mhz, 6 Core(s), 12 Logical Processor(s)
BIOS Version/Date	American Megatrends Inc. 4602, 17/08/2021
SMBIOS Version	3.2

peckerdunn, 2023-03-21:
With the .exe
"C:\Program Files\DrawingBotV3-Premium\DrawingBotV3-Premium.exe"
according to Windows.

peckerdunn, 2023-03-21:
Many thanks.
I'll keep playing around with it in the meantime,

SonarSonic, 2023-11-18:
@peckerdunn Have you encountered this issue with the latest versions?

peckerdunn, 2023-11-18:
Hi. I haven't tried it. I downloaded the latest beta and I'll give it a go in the morning.Thanks for your email.

SonarSonic, 2024-01-29:
[image]

In the latest versions it's possible to bypass OpenCL Acceleration / select the preferred OpenCL Device, if people into this issue in the future, try disabling "Open CL Hardware Acceleration" or try another device.

closed 2024-01-29 by SonarSonic