SonarSonic/DrawingBotV3 #87: Wildcards for Path Finding Controls

simonhavelock, 2023-02-11:
I would like to have the option to do procedural settings for Path Finding Controls. 
Effectively I want to use the Voronoi Stippling for animation purposes (alongside some frames being stippled by AxiDraw) and e.g. would like to have the option to create variable settings for each frame. Say I want to batch process 1000 frames, each called 0000.png, 0001.png, 0002.png, (...) and for each I would like to use %FILE_NAME% wildcard in Point Count (e.g. Point Count = 100*%FiLE_NAME%) which would then create more and more stipples with each frame.

Is this something that could be achieved in DrawingBotV3?

SonarSonic, 2023-02-14:
I'm working on a way to have more automated workflows like this, which would provide even more control to the settings when used like this. It's a little way off however. One way you could achieve this but possibly not too the exact same quality, would be first creating a video where you slowly ramp the brightness from maximum -> nothing, and then run that video though Adaptive Stippling instead of Voronoi Stippling. That would create an effect of increasing stipples with each frame, though it may not be exactly what you're after.

led, 2023-07-22:
To add to the wildcard request, it would be nice if more wildcards (eg keys of the project file etc) were available to use in file naming and also adding comments to generated gcode. As an example, if I re-visit an older generated collection of gcode files, it would be nice to be able to view them and see the context in which they were generated.