SonarSonic/DrawingBotV3 #90: SVG font for symbols

lukehaas, 2023-04-17:
The SVG fonts that have been included with drawingbot are great, but they seem to only have alphanumeric characters.
It would be great if some SVG fonts could be included that had symbols such as mathematical and musical symbols etc.

SonarSonic, 2023-04-17:
At the moment it is possible to add your own fonts by going to Help / Open Config Folder - and copying any SVG fonts you want to use into the "fonts" folder. I will keep this open though as I think I may be able to make this ability clearer in the UI :)

felaschulte, 2023-06-15:
Dear Mr. Landsdell, my DrawingBotV3-Premium doesn‘t work on my new Windows 11 PC. See error Report. What can I do? Greetings Felix Schulte

SonarSonic, 2023-06-16:
>  Dear Mr. Landsdell, my DrawingBotV3-Premium doesn‘t work on my new Windows 11 PC. See error Report. What can I do? Greetings Felix Schulte

Hey Felix! Sorry to hear that, please could you send the crash report too drawingbot@outlook.com and I'll help you get it sorted :)
Best wishes,
Ollie