SonarSonic/DrawingBotV3 #93: Export image as png colour issue

led, 2023-07-26:
**Image File exports in PNG format are 1 bit (B/W)**
No colour appears to be exported when choosing the PNG format, just a B/W version of the render.

**To Reproduce**
Render a plot, 'File->Export per/Drawing->Image FIle, select PNG format, save.

**Expected behavior**
The same colour image as is exported for JPEG, TIFF etc.

**DrawingBotV3 Version**
v1.6.2 Premium

**Operating System**
MacOS

[image]

SonarSonic, 2023-07-27:
Do you have any Blend Mode enabled and do you have "Export Transparent PNGs" enabled?

led, 2023-07-27:
Yes, transparent is set, and I usually have blend mode set (multiply/darkness). Can now confirm that colours are saved when either…

- no blend mode but transparent still set.
- blend mode and transparency is turned off.

closed 2024-07-07 by SonarSonic