SonarSonic/DrawingBotV3 #111: CMYK Notably worse on 1.6.9 Beta vs 1.6.3 Beta

Wyth, 2023-11-26:
Using the same options for adaptive stippling and default CMYK color separation. Only changes made are to the opacity of the pens, and the blending mode to 'darken'.

[image]

SonarSonic, 2023-11-26:
Do you get the same result if you set Clean Black / Clean White to 0?

Wyth, 2023-11-26:
That actually does improve it quite a bit. Setting the Clean Black / Clean White to 0 makes the resulting image much closer to the previous versions, and quite a bit better. Not quite as good (in my opinion as the previous versions, but much more usable now. I'd suggest maybe making the default 0?

closed 2023-11-26 by Wyth

SonarSonic, 2023-11-28:
I will most likely change the defaults, you can also configure the default values in preferences.