SonarSonic/DrawingBotV3 #113: Open line in gcode after export

cojarbi, 2023-11-28:
Im not sure this is a bug but i could note find any setting to adjust this.

As seen on the attachment, after exporting gcode, you can see the beginning/end of path that results in the segment not overlapping. creating a gap.

[image]

SonarSonic, 2023-11-28:
That does look like a bug dealing with closed paths, could you send me the SVG?

cojarbi, 2023-11-28:
Sure, here you go

[image]

SonarSonic, 2023-11-28:
Have you tried plotting with the file as is? I've looked at GCode from DBV3 and there is no issue, the paths are exported as I'd expect. It actually looks like it could be a rendering issue in the other software, or a parsing issue, as it will does the moves needed but is showing them in yellow instead of red.

cojarbi, 2023-11-28:
Yes, please see attached

[image]

SonarSonic, 2023-11-28:
I looks like the plotter needs calibration, most of the lines don't look accurate to me. Some lines seem to bend away before reaching the end, maybe the plotter is running too fast and is slipping.

cojarbi, 2023-11-28:
Noted, after checking, there was a slight difference on squareness. 

Thanks

I will close this one

closed 2023-11-28 by cojarbi