SonarSonic/DrawingBotV3 #118: CL hardware acceleration error on 1.6.11 Beta

jimmykl, 2024-01-15:
**Describe the bug**
CL hardware acceleration (for Streamlines PFMs + Colour match) fails with `The function clCreateCommandQueueWithProperties is not supported` 

**To Reproduce**
Steps to reproduce the behavior:
1. Check CL hardware acceleration is enabled
2. Select a Streamlines PFM or Colour match colour separation
3. Hit 'Start'
4. Drawing hangs with message 

**Expected behavior**
PFM/colour match finishes working, drawing is generated and displayed

**DrawingBotV3 Version**
1.6.11 beta (Premium)

**Operating System**
MacOS 14.2.1 on Apple M1 Max

**Additional info**

Also reported by Tootsie [on Discord](https://discord.com/channels/929089222118359100/929089869949579335/1196462115666546740)

Example log entry
`java.lang.UnsupportedOperationException: The function clCreateCommandQueueWithProperties is not supported
	at org.jocl.CL.clCreateCommandQueueWithPropertiesNative(Native Method)
	at org.jocl.CL.clCreateCommandQueueWithProperties(CL.java:5382)
	at drawingbot.k.c.b.a.a(Unknown Source)
	at drawingbot.k.c.b.f.a(Unknown Source)
	at drawingbot.k.b.c.b.a(Unknown Source)
	at drawingbot.k.b.c.b.a(Unknown Source)
	at drawingbot.k.e.c.b.a(Unknown Source)
	at drawingbot.i.m.c(Unknown Source)
	at drawingbot.i.j.b(Unknown Source)
	at drawingbot.i.j.i(Unknown Source)
	at drawingbot.i.j.call(Unknown Source)
	at javafx.concurrent.Task$TaskCallable.call(Task.java:1426)
	at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)`

SonarSonic, 2024-01-15:
Should be fixed in v1.6.12-beta - let me know if it works :)

jimmykl, 2024-01-15:
Yep, all up and running now 🙏

closed 2024-01-15 by jimmykl