SonarSonic/DrawingBotV3 #119: 1.6.12 CMYK Streamlines Edge Field on Mac M1

TootsieFer, 2024-01-15:
CMYK (regular and extended) fails to plot on a M1 Mac with and without CLA enabled.

**Expected behavior**
It goes through Refining ETF sequence 1-5 fine, but fails to do/show anything when its on Processing - Plotting CMYK sequence.

**Screenshots**
[image]
[image]
[image]
[image]

**DrawingBotV3 Version**
1.6.12

**Operating System**
Ventura 13.5.2

TootsieFer, 2024-01-15:
After I updated to 14.2.1 It was working for me, both either CLA activated or not.

SonarSonic, 2024-01-15:
> For what it's worth CMYK Streamlines Edge Field with CL acceleration _is_ working for me in the 1.6.12 Beta on Macos 14.2.1
> 
> But only after I manually deleted 1.6.11 from Applications then installed the 1.6.12 pkg – might be worth trying? (Or it could be a Ventura thing 🤷)
> 
> ![Cursor_and_Preferences_and_DrawingBotV3_Premium__Version__1_6_12_Beta____DSC4095-Pano-copy-4__and__support___DrawingBotV3_-_Discord_and_Window](https://private-user-images.githubusercontent.com/113574/296866547-19261e71-9714-423e-a261-12393302c1de.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUzNTgwMDEsIm5iZiI6MTcwNTM1NzcwMSwicGF0aCI6Ii8xMTM1NzQvMjk2ODY2NTQ3LTE5MjYxZTcxLTk3MTQtNDIzZS1hMjYxLTEyMzkzMzAyYzFkZS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExNVQyMjI4MjFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NGFkMjVlMGVmY2ZkODM0OWJlMzhjMDQ2ZTAzNzRmYWIyODhmNDNkMTkzMDM5ZDI1YzI5MmZkZDI0NmEzOGJhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.aeZN7ChL1FD6aHyIBa3XVl6372poYTijffBnMLKdi_M)

It looks like in your screenshot that all of the lines are facing the same way, which shouldn't be the case on edge field. Is it doing that all the time with and without CLA?

SonarSonic, 2024-01-15:
> After I updated to 14.2.1 It was working for me, both either CLA activated or not.

Good to know! Are your streamlines all facing the same way?

jimmykl, 2024-01-16:
Good catch @SonarSonic Yes if I turn CL off entirely I get the expected result:

[image]

Note that the previous example had field power 100. With CL on and reducing the power I get the flowfield but it's not interacting with edges:

[image]
ing with the edges

SonarSonic, 2024-01-16:
@jimmykl Could you send me the logs after running it?

jimmykl, 2024-01-16:
Sure thing, here's a fresh log of starting dbv3 (with CL accel on), importing an image, picking cmyk extended and streamlines edge field

[_Log_2024-01-16-14-51-53_0.txt](https://github.com/SonarSonic/DrawingBotV3/files/13944409/_Log_2024-01-16-14-51-53_0.txt)

TootsieFer, 2024-01-16:
@SonarSonic  I just did a check same results as @jimmykl im aslo attaching my log with CLA on
<img width="2051" alt="Screenshot 2024-01-15 at 8 59 24 PM" src="https://github.com/SonarSonic/DrawingBotV3/assets/156711878/d9737bc6-a5f6-
[image]
44f4-b5c3-f816eecb0e6e">
[_Log_2024-01-15-20-54-33_0.txt](https://github.com/SonarSonic/DrawingBotV3/files/13944536/_Log_2024-01-15-20-54-33_0.txt)

TootsieFer, 2024-01-16:
I stumbled upon something interesting: Using the SAME setup and just selecting Apple M1 Max, I am getting a blank even after a significant amount of time compared to the ones with  either OFF or AUTO .
[image]

SonarSonic, 2024-02-02:
@TootsieFer @jimmykl Looking into this issue again, does Colour Match create the same results with CLA on/off?

jimmykl, 2024-02-08:
@SonarSonic Excuse delay, I'm getting weird bands straight lines using Colour Match with CLA on…

[image]

grinchdubs, 2024-02-16:
I also just recently ran into the issue with the openCL check box being on and only getting straight edge flow lines,once turned off everything was working as intended but taking much longer

closed 2024-04-11 by SonarSonic