SonarSonic/DrawingBotV3 #121: HPGL ESC commands

ilachEU, 2024-02-22:
**Is your feature request related to a problem? Please describe.**
My HPGL plotter requires sending several ESC commands before the actual drawing to work correctly with DrawingBotV3 

**Describe the solution you'd like**
In the HPGL setup window, add the ability to set the HPGL start sequence, including the ability to specify ESC characters

**Describe alternatives you've considered**
I now have a separately made HPGL file with the start sequence that I have to send to plotter for the actual drawing

SonarSonic, 2024-02-23:
I will introduce this for you in a future version as I've received similar requests. What HPGL plotter is it out of interest?

ilachEU, 2024-02-23:
Aritma Colorgraf 0512

SonarSonic, 2024-04-15:
Added in v1.6.14

closed 2024-04-15 by SonarSonic