SonarSonic/DrawingBotV3 #124: Menus and modals do not render properly on Ubuntu Regolith (i3 tiling window manager)

falko-apheris, 2024-03-04:
**Describe the bug**
- The top menu bar can not be opened with a mouse since the dropdown appears like 20 px below the button and disappears immediately when the mouse moves into the gap between the button and the dropdown.
- modals appear very small in height and can not be resized

see attached images 

On the regular Ubuntu desktop manager everything works just fine.

**To Reproduce**
Steps to reproduce the behavior:
Start the app in a i3 tiling window manager.

**Expected behavior**
Menus and modals render like on regular Ubuntu desktop manager

**Screenshots**
[image]
[image]

**DrawingBotV3 Version**
v1.6.13-stable-free

**Operating System**
Ubuntu 22.04.4 LTS with Regolith 2 https://regolith-desktop.com/

falko-apheris, 2024-03-04:
https://www.reddit.com/r/i3wm/comments/jzciw9/wrong_menu_position_on_any_javafx_application/ this is exactly the problem
I have the feeling the modal issue could be fixed by allowing resizing of the modal.

SonarSonic, 2024-03-08:
The JDK team have pushed a fix to a newer version of JavaFX, I will be update to that and hopefully it will be resolved! https://bugs.openjdk.org/browse/JDK-8251240

select, 2024-03-15:
I tried it on the new alpha. It's solve :heart:

falko-apheris, 2024-03-18:
[image]

closed 2024-03-18 by falko-apheris