SonarSonic/DrawingBotV3 #131: Pen preset dialog problem

led, 2024-03-08:
When creating or editing a pen preset, the dialog that opens is small and appears to contain no elements.

Steps to reproduce the behavior:
1. Go to the 'Pen Settings' panel and hit the 'Presets' button.
2. Select 'New Preset' (or Edit etc)
3. A dialog opens - this is the problem.

**Expected behavior**
Expected to be able to manage (eg create/update) preset attributes (name etc).

[image]

The dialog (on the right) is what currently appears.

**DrawingBotV3 Version**
1.6.143 Alpha

**Operating System**
Mac M1

SonarSonic, 2024-03-11:
I can't recreate this behaviour at the moment, can you replicate it consistently? Could you maybe send me your logs from "Help/Export Logs & Crash Reports"

led, 2024-04-11:
As in the Discord thread…

> Can you also let me know can you edit the presets if you open File/Preset Manager, and also does this issue happen everytime, or only sometimes?

If I open DBv3 and access the presets before loading an image, then the preset window opens as it should do. If I've loaded an image, then I just get the narrow empty window as screenshotted. This happens every time. 

Log attached.

[drawingbotv3 premium_logs_mac_2024-04-11-11-57-41 2.zip](https://github.com/SonarSonic/DrawingBotV3/files/14945263/drawingbotv3.premium_logs_mac_2024-04-11-11-57-41.2.zip)

matsylmatsyl, 2024-04-11:
Thanks for the response. I see how you can open the preset manager before
loading an image. However, I don't think this helps (me) very much. Unless
I am missing something it looks like you would have to create a preset
(using this method) without being able to preview how it works. For
instance, every preset I have saved is starting from or based on another
preset. I load an image and run a preset, make adjustments and then save
the new preset. But, of course this won't work (now) if an image has been
loaded. The only way I can see using the method you described would be to
open an image, apply a preset, adjust things until you get something you
want to save as a preset. At that point I guess you could copy all of the
settings, close DB, open it again (without loading an image), open the
preset window and then copy all of the settings you have written down to
the preset window and then save the preset at that point. I guess that
would work, but it's a real pain. Again, maybe I don't understand your
reply, but I think this is a big problem for me unless it can get
corrected. I use DB a lot (with many custom presets) and all of a sudden
this problem cropped up-something must have happened. I really hope Ollie
can figure it out-he's brilliant-I have my fingers crossed. best wishes...

led, 2024-04-12:
Can confirm that this appears to be fixed. Thanks.

closed 2024-04-12 by led

matsylmatsyl, 2024-04-13:
Fixed for me as well. Thanks for responding so quickly.