SonarSonic/DrawingBotV3 #132: Wrong desktop icon

select, 2024-03-08:
**Describe the bug**
When you search for the software some generic java icon is shown. Only when you start the software the proper DB icon (maybe you want to choose something prettier in the future, since this is an app for design ;-) is shown. See screenshot

**To Reproduce**
Search for the app, look at the icon.
Start the app look at the icon

**Expected behavior**
App icon is consistent

**Screenshots**
[image]

**DrawingBotV3 Version**
DrawingBotV3-Premium-1.6.13-stable-linux.deb

**Operating System**
Ubuntu 22.04.4 LTS

**Additional info**
(Add any other info about the problem here.)

andypiper, 2025-03-19:
The same issue exists on Fedora 41. 

```
$ cat /opt/drawingbotv3/lib/drawingbotv3-DrawingBotV3-Premium.desktop
[Desktop Entry]
Name=DrawingBotV3-Premium
Comment=DrawingBotV3-Premium
Exec=/opt/drawingbotv3/bin/DrawingBotV3-Premium
Icon=/opt/drawingbotv3/lib/DrawingBotV3-Premium.png
Terminal=false
Type=Application
Categories=Unknown
MimeType=text/plain
```

The file /opt/drawingbotv3/lib/DrawingBotV3-Premium.png is a Java applet logo / icon (as shown in the issue). Simply replacing that with the icon file from this repository improved things for me.

I rewrote the .desktop file:

```
[Desktop Entry]
Name=DrawingBotV3-Premium
Comment=Converts images into stylized drawings for pen plotter art, digital artwork, image/video stylisation, NFT creation, embroidery, and graphic desi
gn.
Exec=/opt/drawingbotv3/bin/DrawingBotV3-Premium
Icon=/opt/drawingbotv3/lib/DrawingBotV3-Premium.png
Terminal=false
Type=Application
Categories=Graphics;Utility;
MimeType=application/x-drawingbotv3;image/tiff;image/x-tga;image/png;image/jpeg;image/gif;image/bmp;image/svg+xml;
StartupWMClass=drawingbot.c
```

I also updated the mime type file to set the project files as `application/x-drawingbotv3`

At the moment, I don't think there is much more that can be done to improve the behaviour given the relative lack of movement on JavaFX + Wayland (I see it currently shows up on my desktop via XWayland, rather than directly), but that at least gets things to a slightly nicer position - fixes the icon, puts DrawingBotV3 in a more logical location in the menus, improves the description, and attempts to connect the running app to the window.

SonarSonic, 2025-07-06:
Thanks both for your suggestions, I've now fixed the appearance of the icon + made some changes to the .desktop file as you suggested @andypiper - note on Fedora Linux when launched it appears the application icon is still not being used properly, perhaps this will need another issue

closed 2025-07-06 by SonarSonic