SonarSonic/DrawingBotV3 #139: Mosaic Custom doesn't pick up drawing sets correctly.

led, 2024-07-05:
**Describe the bug**
When using the Mosaic Custom PFM, it's pretty much impossible to select different drawing sets when you have multiple masked areas as the Drawing Sets drop-down isn't populated correctly.

**To Reproduce**
Set up the 'Mosaic Custom' PFM. Add in a mask image and try to select different drawing sets for the different masked colours. For me, the drop-down with the Drawing Sets isn't correctly populated.

**Expected behavior**
To be able to select different drawing sets that have been created.

**DrawingBotV3 Version**
1.6.16 Stable

**Operating System**
Mac M2

SonarSonic, 2024-11-14:
[image]

Testing just now at it seems to be working, could you provide a screenshot of what you're seeing if you're still encountering this issue?

SonarSonic, 2025-07-06:
Closing as not able to reproduce this issue currently