SonarSonic/DrawingBotV3 #149: Bug: Incompatibility issue with Fedora Linux, missing ocl-icd-opencl-dev package

SonarSonic, 2024-09-04:


andypiper, 2025-03-17:
FWIW I have installed DrawingBotV3-Premium-1.6.21-stable-linux-x86-64.rpm on Fedora 41. 

It complained about needing ocl-icd-opencl-dev which is not available. 

There is a package ocl-icd-devel which I installed, and then installed the DB rpm with --nodeps and it seems to start up ok.

I suspect this is a difference in package naming between Debian and Red Hat packaging styles. RPMs are typically -devel and on the DEB side, there's more often a -dev package.

andypiper, 2025-03-19:
I don't see where the ocl-icd-opencl-dev dependency exists in the configuration files, or I'd have a look at how to fix this. I assume (from a bit of digging into the repo) that this is mostly all handled by jpackage, but the dependencies must be listed somewhere that I'm not seeing.

The same applies to some of the elements around the .desktop file integration (#132) which I was looking at - I'm guessing this is also somehow configured via jpackage setup.

SonarSonic, 2025-03-20:
@andypiper Thanks for looking into it, as OpenCL is only needed in the Premium version, the Linux dependencies are not in the build.gradle on this repository. I'll look into it, and try building the rpm with ocl-icd-devel instead.

andypiper, 2025-03-20:
Makes sense, thanks!