SonarSonic/DrawingBotV3 #152: Batch Processing 'sourceImage' is null

paul-ferragut, 2024-11-13:
**Describe the bug**
batch processing not working anymore
'Batch processing - java.lang.NullPointerException: Cannot invoke 'java.awt.image.BufferedImage.getWidth()' because 'sourceImage' is null
I have a folder with png images

**To Reproduce**
Steps to reproduce the behavior:
1. Link input and output folder
2. Click on 'Start Batch Processing'

**Expected behavior**
Process images and export

**Screenshots**

**DrawingBotV3 Version**
v1.6.20-Premium

**Operating System**
Windows

[image]

closed 2024-11-13 by SonarSonic

SonarSonic, 2024-11-14:
Fixed in v1.6.21-stable, which is out now!

paul-ferragut, 2024-11-18:
It works, Thanks!!