SonarSonic/DrawingBotV3 #153: Current document is unable to create an element of the requested type . . .

irwin613, 2024-12-05:
I created a python script that executed and produced the desired result.  The output is a multicolored graphic with each colour(5) saved as its own layer in an Inkscape-optimised SVG file.  I also tried opening the file in a 'regular' SVG format but still get the same error message.

The complete error message is namespace:http://www.w3.org/2000/svg, name:plotdata

[inspiral.docx](https://github.com/user-attachments/files/18029000/inspiral.docx)

SonarSonic, 2024-12-05:
Could you upload the SVG here to?

irwin613, 2024-12-05:
As requested

irwin613, 2024-12-06:
[image]

irwin613, 2024-12-09:
@Ollie  regarding issue #153.  Playing around with the script for this issue, I removed the requirement to save each color as a separate layer within the svg file.  The output, as modified is readable by DBv3.  Like the original, it is also readable by Inkscape.  the only difference being the layer structure in Inkscape makes it more difficult to plot out.

I can now modify the output using your program, but the output is more finiky to manipulate when it comes to plotting out via Inkscape.  Is there a fix that can be implemented in DBv3 so that the flexibility of separate layers within an SVG file can be maintained?

ollie, 2024-12-10:
@irwin613 I believe you have mentioned the wrong ollie. :)

irwin613, 2024-12-10:
Haha, thanks!

SonarSonic, 2024-12-27:
@irwin613 Is each layer just a different colour, in which case it can be maintained by exporting as an Inkscape SVG, but if each layer contains multiple colours then at the moment this is not possible.

irwin613, 2024-12-27:
@SonarSonic   Each color is indeed a separate layer. I suppose that l can export each layer as a separate file but that gets to be tedious and then there is the concern about ensuring each new layer is in register with previous ones. I should mention that this error doesn’t always occur. I have numerous svg files where colors are saved as individual layers within one file,
Without that error message. Thanks for trying.

SonarSonic, 2025-03-19:
I've tweaked how SVGs are loaded to handle invalid elements, tested with this SVG and it's working now. Will be fixed in v1.6.22-stable

closed 2025-03-19 by SonarSonic